adabru/BleWinrtDll #6: Disconnected peripherals, Reading specific characteristics.

Quget, 2021-01-21:
Hello! 

I started using this library today in my Unity Project, seems nice to use so far.
I am able to scan, connect(trough 'polling' services), subscribe and write to specific characteristics 👍 

But I did end up with 2 problems.

I can't figure out how to see if my peripheral got disconnected. 
This is quite handy to know if I want to automaticly reconnect to the device.

I also can't figure out how to read a specific characteristic.

adabru, 2021-01-21:
Hi @quget,

checking the connection status and requesting a value from the device is not implemented as I didn't had the need for that yet.

For the connection status, I guess you can try using a [GattSession](docs.microsoft.com/en-us/uwp/api/windows.devices.bluetooth.genericattributeprofile.gattsession) with `MaintainConnection` set to true, as it is mentioned in the note on [this doc page](https://docs.microsoft.com/en-us/windows/uwp/devices-sensors/gatt-client). The session has also an event in the case the connection-status changes.

Reading/Requesting a specific characteristic value from the device can be done with the [ReadValueAsync function](docs.microsoft.com/en-us/uwp/api/windows.devices.bluetooth.genericattributeprofile.gattcharacteristic.readvalueasync) . You can use the code of `SendData` as template for that. This is the easier one. Alternatively you can write to one characterstic and program your device so that it will trigger a notification of the characteristic you're interested in.

Quget, 2021-01-22:
Thanks for the quick reply! I did try to go trough the source code.
But I don't know c++ so well. It will take me a while to understand it.

adabru, 2021-01-22:
You can checkout the [Microsoft sample (C# and equivalent C++ code)](https://docs.microsoft.com/en-us/samples/microsoft/windows-universal-samples/bluetoothle/) as reference for ReadValueAsync. 

In my case I first got it to work in C# UWP, then in a C++ winrt console program and then in the dll. If you'll have specific questions regarding the code, feel free to ask.

Quget, 2021-01-22:
I think I manage to get the read one in.  Just got to test it out.
Using the same queue as the subscribed characteristic.

Connection status will be a bit of a problem but I will get there I think :)

adabru, 2023-03-22:
As reference for reading specific characteristics:

tufeixp's ReadDataAsync function: https://github.com/adabru/BleWinrtDll/issues/30#issuecomment-1118295341

SendData code: https://github.com/adabru/BleWinrtDll/blob/f2635b7b02432ea3006983657107f092c862171d/BleWinrtDll/BleWinrtDll.cpp#L528-L561