adabru/BleWinrtDll #7: BleWinrtDll.cpp:501 SubscribeCharacteristicAsync catch: 属性需要进行身份验证，然后才能读取或写入。

Jason-Chn95, 2021-02-01:
May I ask if this is the reason why the sensor needs authentication or something else? This is the first time for me to do this Bluetooth function.

Jason-Chn95, 2021-02-01:
[image]

adabru, 2021-02-01:
I didn't do any bluetooth authentication yet. From the error message I'd also guess you need it. 

Maybe pairing it will help: 
<https://docs.microsoft.com/en-us/windows/uwp/devices-sensors/bluetooth-dev-faq#why-does-my-bluetooth-le-device-stop-responding-after-a-disconnect> . You could also try the microsoft sample <https://github.com/microsoft/Windows-universal-samples/tree/master/Samples/BluetoothLE> and see whether you can read the value without authentication instead of subscribing to it. You can also use the microsoft sample to try out the pairing code.

Jason-Chn95, 2021-02-02:
> I didn't do any bluetooth authentication yet. From the error message I'd also guess you need it.
>
> Maybe pairing it will help: [...]

Thanks adabru for your help,now I solved this problem.I have another question. Do you know how to do it?I want to connect two Bluetooth devices on my PC,how to do this ?

adabru, 2021-02-02:
I guess each bluetooth device normally has a unique device id. In that case you should be able to just call the functions with different device-ids. Each received data-package contains information about the device it was sent from. Likewise, in each data-package you send you can specify the device-id it should go to.

Jason-Chn95, 2021-02-03:
> 我猜每个蓝牙设备通常都有一个唯一的设备ID。在这种情况下，您应该能够仅使用不同的设备ID来调用函数。每个接收到的数据包都包含有关其发送设备的信息。同样，在您发送的每个数据包中，您都可以指定设备编号。

Thank you. I can only send you a star

closed 2021-02-03 by Jason-Chn95

Augmentaio, 2021-04-26:
Hi Jason, did you face this issue when testing with beacons? How did you solve it?

Jason-Chn95, 2021-04-27:
> Hi Jason, did you face this issue (SubscribeCharacteristicAsync catch) when testing with beacons? How did you solve it?

Sorry,I didn't have that problem