adabru/BleWinrtDll #8: Why can Bluetooth module on PC connect to Bluetooth device, but can't refresh to Bluetooth device in demo？

Jason-Chn95, 2021-02-09:
Hello adabru, you can't believe I'm having another problem and I need your help. I can connect to Bluetooth device on the Bluetooth module of PC, but I can't search any Bluetooth device by clicking scan devices in the program. Do you know what happened?

adabru, 2021-02-09:
What kind of device is it?
Are other BLE-devices shown in the scan listing?
Did it work before?

Jason-Chn95, 2021-02-10:
> What kind of device is it?
> Are other BLE-devices shown in the scan listing?
> Did it work before?

devices worked in other pc,when I test in my pc it worked,but in Lenovo Zen7000 the devices couldn’t be found.
I have print the log in exe,it shows the Winrtdll
missing?
Here is the log
--------------------------------------
2021/2/10 13:12:13:Exception: 
condition: DllNotFoundException: BleWinrtDll.dll
stackTrace: Demo.Update () (at <96e8b8da99344dfc86d5372a27b8f721>:0)

adabru, 2021-02-10:
Yeah, BleWinrtDll.dll should not be missing. Are you running the demo from the Unity Editor or from a Windows Standalone Export? The BleWinrtDll.dll should be right in the Assets folder, could you check that?

Jason-Chn95, 2021-02-12:
> 是的，BleWinrtDll.dll应该不会丢失。您是从Unity编辑器还是从Windows Standalone导出运行演示？BleWinrtDll.dll应该位于Assets文件夹中，您可以检查一下吗？

Yes, I export it to run as exe. I can find it in the plugins folder BLEWinrt.dll , but the running log shows missing, which is strange.

adabru, 2021-02-12:
When I tried to run it on another computer than my development computer, I had to install the visual studio redistributable on that machine. But the error message also told `vcredisxx not found` or something like that. If you're running it on another machine, maybe that one could be missing? Also, did you look whether the dll is also located in the exported folder?

Jason-Chn95, 2021-02-18:
> Visual Studio

OK, I will try to open the service in windows and install visual studio. The exported DLL exists in the plugins folder

adabru, 2021-02-18:
It is sufficient to install the visual studio redistributable: <https://support.microsoft.com/en-us/topic/the-latest-supported-visual-c-downloads-2647da03-1eea-4433-9aff-95f26a218cc0>

Jason-Chn95, 2021-02-19:
> 安装Visual Studio可重新分发就足够了：[https](https://support.microsoft.com/en-us/topic/the-latest-supported-visual-c-downloads-2647da03-1eea-4433-9aff-95f26a218cc0) : [//support.microsoft.com/zh-cn/topic/the-latest-supported-visual-c-downloads-2647da03-1eea-4433-9aff-95f26a218cc0](https://support.microsoft.com/en-us/topic/the-latest-supported-visual-c-downloads-2647da03-1eea-4433-9aff-95f26a218cc0)

Thanks very much!!!