adabru/BleWinrtDll #9: Subscribe never finishes

brunocoimbrar, 2021-02-25:
I have a custom device that I am trying to interact with and using another plugin (for android and ios) I got it to work nicely ( https://assetstore.unity.com/packages/tools/network/bluetooth-le-for-ios-tvos-and-android-26661 ) but now I am trying to get support for Windows too (and especially inside the Unity Editor for quick prototyping).

On the demo scene after selecting my device, the correct service, and the correct characteristic, hitting Subscribe does nothing (by "does nothing" I mean that "if I change it to "block: true" it just freezes the demo scene).

Any idea here?

adabru, 2021-04-15:
To further investigate you can install Visual Studio Community with C++ Desktop and UWP components., modify the UUIDs in `DebugBle/Program.cs` and run the `DebugBle` program.

If it also freezes there, you can download and run the [Microsoft's Bluetooth Low Energy sample](https://docs.microsoft.com/en-us/samples/microsoft/windows-universal-samples/bluetoothle/) and see whether that works out. If the demo works, it should be fixable in this repo.

jadware, 2021-10-04:
looking at Demo.cs line 128:

`if (isSubscribed)
{
    BleApi.BLEData res = new BleApi.BLEData();
    while (BleApi.PollData(out res, false))
    {
        subcribeText.text = BitConverter.ToString(res.buf, 0, res.size);
        // subcribeText.text = Encoding.ASCII.GetString(res.buf, 0, res.size);
    }
}`

This looks like the app could easily get stuck in an infinite loop (on the main thread) if PollData is doing something unexpected like trying to deal with subscribing to a characteristic that isn't subscribable.  Guessing the error is here!

adabru, 2022-08-14:
This could have been fixed by https://github.com/adabru/BleWinrtDll/pull/25.