adabru/BleWinrtDll #10: Any idea how to add RSSI / signal strength?

jadware, 2021-02-27:
It is working well for me, thank you.  How might it be possible to get RSSI signal strength of the devices that are found in the scan?  There may be a property System.Devices.Aep.SignalStrength that can be added to the requested properties list inside StartDeviceScan, but I am not sure how to retrieve its value afterwards.

Much appreciate your help if possible!

adabru, 2021-04-15:
Your approach seems good to me.

Add the `System.Devices.Aep.SignalStrength` property to the requestedProperties in [BleWinrtDll.cpp#L268](https://github.com/adabru/BleWinrtDll/blob/main/BleWinrtDll/BleWinrtDll.cpp#L268)

Add a `int32_t signalStrength;` property to the `DeviceUpdate` struct at [BleWinrtDll.h#L10](https://github.com/adabru/BleWinrtDll/blob/main/BleWinrtDll/BleWinrtDll.h#L10).

Populate the field at [DeviceWatcher_Updated](https://github.com/adabru/BleWinrtDll/blob/main/BleWinrtDll/BleWinrtDll.cpp#L244) and [DeviceWatcher_Added](https://github.com/adabru/BleWinrtDll/blob/main/BleWinrtDll/BleWinrtDll.cpp#L227) with:	
```c++
if (deviceInfoUpdate.Properties().HasKey(L"System.Devices.Aep.SignalStrength")) {
	deviceUpdate.signalStrength = unbox_value<int32_t>(deviceInfoUpdate.Properties().Lookup(L"System.Devices.Aep.SignalStrength"));
}
```

Add a corresponding C# property at [BleApi.cs#L26](https://github.com/adabru/BleWinrtDll/blob/main/BleWinrtDll%20Unity/Assets/BleApi.cs#L26):
```C#
        [MarshalAs(UnmanagedType.I4)]
        public int nameUpdated;
```

For debugging you can use the `DebugBle` cli-program or the [Microsoft's Bluetooth Low Energy sample](https://docs.microsoft.com/en-us/samples/microsoft/windows-universal-samples/bluetoothle/).

jadware, 2021-04-15:
thanks!!  working