adabru/BleWinrtDll #11: Unity crash when I press the subscribe button(MRTK)

kisstherain8677, 2021-04-15:
Hello!

I am in a project of AR and using the MRTK UGUI to conncet to my BLE device. I almost paste your 2D demo script on my project. I can get the correct device, service and characteristics. However, when I press the subscribe button, the unity just crashes. Do you know how to solve this issue? 
[image]

adabru, 2021-04-15:
To further investigate you can install Visual Studio Community with C++ Desktop and UWP components., modify the UUIDs in `DebugBle/Program.cs` and run the `DebugBle` program.

If it also crashes there, you can download and run the [Microsoft's Bluetooth Low Energy sample](https://docs.microsoft.com/en-us/samples/microsoft/windows-universal-samples/bluetoothle/) and see whether that works out. If the demo works, it should be fixable in this repo.

kisstherain8677, 2021-04-15:
#

> To further investigate you can install Visual Studio Community with C++ Desktop and UWP components., modify the UUIDs in `DebugBle/Program.cs` and run the `DebugBle` program.
>
> If it also crashes [...]

Sry, I think you misunderstand my issue. Your demo works just fine(I can conncet to my BLE device and receive data). The problem is that I copy your script to a MRTK UGUI, and though I can get the correct device, service and characteristics UUIDs, my unity crashes while I press the Subscribe button(Actually, it crashes while it is running `SubscribeCharacteristic(string deviceId, string serviceId, string characteristicId, bool block)`). However, I change nothing in the script, so I wonder if the issue is that the code in the BleWinrtDll Unity\Assets\BleWinrtDll.dll cannot be applied in MRTK UGUI?

adabru, 2021-04-15:
Thanks for correcting me! It could theoretically be that not all UWP-functionality is available on the mixed reality platform or that it works subtly differently. I'm not familiar with the mixed reality platform but according to [this microsoft article](https://docs.microsoft.com/en-us/windows/mixed-reality/develop/porting-apps/building-2d-apps), one should be able to run a UWP application on the mixed reality headset:

> Windows Mixed Reality lets your users see holograms as if they're right around them in the physical and digital world. At its core, both HoloLens and the Desktop PCs you attach immersive headset accessories to are Windows 10 devices. You're able to run almost all Universal Windows Platform (UWP) apps in the Store as 2D apps.

I guess to check out whether the UWP-BLE-Api, more specifically the subscribe function, is applicable on the platform one could compile Microsoft's Bluetooth Low Energy sample as UWP-app and run it on the headset.

If scanning services and characteristics works, the only additional UWP/winrt functions that are called for subscription are `characteristic.WriteClientCharacteristicConfigurationDescriptorAsync` and `characteristic.ValueChanged` ([BleWinrtDll.cpp#L482-L504](https://github.com/adabru/BleWinrtDll/blob/main/BleWinrtDll/BleWinrtDll.cpp#L482-L504)).

kisstherain8677, 2021-04-16:
Thanks for your explaination! I tried to change the type of function `
bool SubscribeCharacteristic(wchar_t* deviceId, wchar_t* serviceId, wchar_t* characteristicId, bool block) `from bool to void,and it works now.

monkeyfuckeddog, 2022-10-14:
> 谢谢你的解释！我试图将函数的类型`bool SubscribeCharacteristic(wchar_t* deviceId, wchar_t* serviceId, wchar_t* characteristicId, bool block)`从 bool 更改为 void，它现在可以工作了。

Thanks for sharing!