adabru/BleWinrtDll #12: ANDROID support

michaelSOLERbeatty, 2021-05-14:
There is a way to have android support to this asset? Thanks

adabru, 2021-05-14:
UWP (≈Winrt) is not Android's friend it seems. I'm sorry, I don't know whether there is a framework where one code runs on both Windows Standalone and on Android. At least not for free.

I did a google search and found:
- Asset Store: [Bluetooth LE for iOS, tvOS and Android](https://assetstore.unity.com/packages/tools/network/bluetooth-le-for-ios-tvos-and-android-26661)
- Asset Store: [Arduino Bluetooth Plugin](https://assetstore.unity.com/packages/tools/input-management/arduino-bluetooth-plugin-98960)
- Asset Store: [Bluetooth Android Plugin](https://assetstore.unity.com/packages/tools/network/bluetooth-android-plugin-15179)
- <https://github.com/seiji/unity-bluetooth>: actually that one looks like a project that could potentially merge this project, so you would have Windows Standalone, Android and iOS support
- <https://github.com/elmirjagudin/UnityBluetooth>

Feel free to add other ble for android frameworks/wrappers or recommend specific ones.

adabru, 2023-01-22:
There's another active open source project: https://github.com/Velorexe/Unity-Android-Bluetooth-Low-Energy