adabru/BleWinrtDll #13: Unity skipping .Net 5

nathanrona, 2021-05-23:
FYI, Unity is not planning to implement .Net 5, rather they plan to skip to .Net 6 
_".NET 6 support: Peterson described this as "still a bit fuzzy," with the company hoping to provide a preview around the time the landmark, unifying, milestone .NET 6 ships in November as a long-term support (LTS) release, meaning that "Unity will likely skip .NET 5 entirely." He described the biggest hurdles for this as:
This is a breaking change for all Unity users. Any assemblies compiled against mscorlib.dll from the .NET Framework ecosystem will not work, and must be recompiled.
The .NET Core ecosystem does not support domain reloading, which is a fundamental requirement of the Unity Editor. We have done work to use Assembly Load Context instead, but that will be a breaking change, and will require time to implement properly, both from the Unity side and in user projects. "_

https://visualstudiomagazine.com/articles/2021/04/22/unity-plans.aspx

adabru, 2021-05-23:
Thanks for reporting! There could be another thing delaying first-party C#-support for windows' BLE-api. It is not sure whether Unity will support this part of .NET 6 api. If someone is motivated to find out and let the others know you can follow the instructions in [this post](https://forum.unity.com/threads/unity-future-net-development-status.1092205/page-3#post-7090225).

Also feel free to create a pull request to update the readme.

adabru, 2022-03-20:
As Abrahamh08 pointed out in https://github.com/adabru/BleWinrtDll/issues/27 , .Net 5 wouldn't improve the situation with BLE.