adabru/BleWinrtDll #15: Would you be able to implement BluetoothLEAdvertisementWatcher, connection status ?

nathanrona, 2021-05-27:
Thank you very much for this great API. Thans to it was able to connect to BLE sensor and subscipe to notification, however I woud want to: 
1. Check connection status, i.e. to know if I am connected and subsciption is active. 
2.  Scan only for devices that support my desired service (cycle cadance and speed service) 

For the later I need to use BluetoothLEAdvertisementWatcher instead of devicewatcher. 
For the later I'm not sure, in Web-bluetooth (JS) I set an iternal connected varible true upon completing the pairing and set it false upon disconnect event. 

Looked at the code in hope to be able to do it myself, but I'm not familiar in C++.

adabru, 2021-05-27:
For the connection status, I guess you can try using a [GattSession](docs.microsoft.com/en-us/uwp/api/windows.devices.bluetooth.genericattributeprofile.gattsession) with `MaintainConnection` set to true, as it is mentioned in the note on [this doc page](https://docs.microsoft.com/en-us/windows/uwp/devices-sensors/gatt-client). The session has also an event in the case the connection-status changes. There already was a need for it in https://github.com/adabru/BleWinrtDll/issues/6 , but no implementation yet. 

That's for the connection. Subscription is read and written (="subscribe") via the ClientCharacteristicConfigurationDescriptor, see https://github.com/adabru/BleWinrtDll/blob/main/BleWinrtDll/BleWinrtDll.cpp#L486 or [ReadClientCharacteristicConfigurationDescriptorAsync](https://docs.microsoft.com/en-us/uwp/api/windows.devices.bluetooth.genericattributeprofile.gattcharacteristic.readclientcharacteristicconfigurationdescriptorasync). It could be that your bluetooth device would already hold the subscription with `MaintainConnection`. Maybe it could require one to periodically communicate, i.e. read or write some arbitrary value.

BluetoothLEAdvertisementWatcher seems good if you say so.

If you're able to create a working C#-sample I could probably help you translating it into the winrt-version. If you don't have a C# sample to start with you can use the [Microsoft sample (C# and equivalent C++ code)](https://docs.microsoft.com/en-us/samples/microsoft/windows-universal-samples/bluetoothle/) as a starting point.

nathanrona, 2021-06-02:
Found an example of BluetoothLEAdvertisementWatcher  here https://github.com/urish/win-ble-cpp