adabru/BleWinrtDll #16: Nuget Package  Microsoft.Windows.CppWinRT needed to build solution

nathanrona, 2021-06-02:
Cloned project and tried to build according to instructions in Reame.md, however build failed on error "wait_for is not a member of winrt::impl" 

Found [solution ](https://github.com/microsoft/Windows.UI.Composition-Win32-Samples/issues/47) being installing the Nuget Package Microsoft.Windows.CppWinRT,
1. Right-click on Refernces in solution window. 
2. Click manage Nuget packages...
3. Click browse tab
4.  Find Microsoft.Windows.CppWinRT
5. Install 
After that building worked.

adabru, 2021-06-02:
Thanks for sharing!

nathanrona, 2022-01-19:
Thanks, but still has issue 	Cannot open include file: 'winrt/Windows.Foundation.Collections.h': No such file or directory	BleWinrtDll	C:\Users\n2\Documents\GitHub\BleWinrtDll\BleWinrtDll\stdafx.h

adabru, 2022-02-09:
If someone could create a pull request that fixes this issue, that would be highly appreciated!

monkeyfuckeddog, 2022-08-23:
[image]

After this, still have error posted.

adabru, 2024-06-08:
I found someone who fixed it: https://github.com/FeuFeve/BleWinrtDll/commit/7a6ba261ec807325c7eee124a609aaba4e377802

If someone uses the fix, feel free to create a pull request. The commit above has some other changes that may need to go to the .gitignore?