adabru/BleWinrtDll #19: BleWinrtDll.cpp:501 SubscribeCharacteristicAsync catch: The attribute cannot be written.

qwetan987, 2021-08-10:
hi when i press subscribe it returns this error

is there any solution for this

thanks

adabru, 2021-08-14:
For investigation, you can download and run the [Microsoft's Bluetooth Low Energy sample](https://docs.microsoft.com/en-us/samples/microsoft/windows-universal-samples/bluetoothle/) and see whether subscribing works there. If the demo works, it should be fixable in this repo.

Did you already try subscribing with another platform/program?

PatrykSajdok, 2021-08-17:
I was able to subscribe in Microsoft's sample, but I have the same error in Unity

adabru, 2021-08-17:
Does this mean in Microsoft's sample the subscription failed with the same error?

PatrykSajdok, 2021-08-18:
I had problem in Unity (The attribute cannot be written), then I've checked Microsoft's sample and everything was working there and after that for some reason error in Unity disappeared. However I still cannot get characteristc values from some BT devices.