adabru/BleWinrtDll #20: Unable to write to device

QuackyQuack, 2021-08-10:
Hiya,

I recently started use this library and It is amazing, thank you very much.

I am trying to get this working in the Unity Editor for faster debugging. After connecting to my BLE device I am able to receive data just fine but the device does not receive any data I try to send.

I know the device works because I can send data through our webpage.

I am not too familiar with Bluetooth development so I was wondering if anyone had run into this issue. I was also wondering what the best way I can debug this issue is?

Thank you.

adabru, 2021-08-14:
For investigation, you can download and run the [Microsoft's Bluetooth Low Energy sample](https://docs.microsoft.com/en-us/samples/microsoft/windows-universal-samples/bluetoothle/) and see whether writing the characteristic works there.

Can you see any error message in the GUI?

You could also try out changing `GattWriteOption::WriteWithoutResponse` to `GattWriteOption::WriteWithResponse` at https://github.com/adabru/BleWinrtDll/blob/main/BleWinrtDll/BleWinrtDll.cpp#L536 . That would make the call wait for the BLE device to send a "I received" confirmation. In BLE terms that would be "request"+"response" instead of a "command" without a response.

MikEfraim, 2021-12-11:
Hello,

I managed to compile the code and changed the line 536. I customized the code and I'm trying to write a letter "C" or "F" but when I press the Write Button I get:

"BleWinrtDll.cpp:545 SendDataAsync catch: The attribute cannot be written."

I also seem to subscribe to the first characteristic of my BLE peripheral despite clicking on the second characteristic.

This bug only happens from within unity, it doesn't happen with either the BLE scanner app from my android phone nor the same PC BLE scanner app. Those apps seem to work fine.

ehauner, 2022-02-04:
> if I have only one characteristic the write button works fine

Thank you. This hint helped me solve this issue. I logged a bunch of things and found out that changing the characteristics dropdown was not correctly setting the selected characteristic in the script. It was always trying to subscribe to the first one since the script selects the first item by default.

This issue is that the dropdown was never hooked up to the SelectCharacteristic function in the Demo script.

Simple fix:
1. select the characteristics dropdown in unity
2. in the inspector, find "On Value Changed (Int32)" in the "Dropdown" component
3. add Demo.SelectCharacteristic and pass "DropdownCharacteristic" as the argument

adabru, 2022-02-05:
@ehauner would you mind creating a pull request :) ? Currently I don't have the setup to test this.

tommasoFabbri, 2023-12-15:
Hi,
we started to use the BlewinrtDll we need it for our Unity app and the solution seems great.

we can connect to the BLE device and can receive data but the function "SendData" always return false 

we have try to use the precompiled dll that e find in the unity example non threaded
the device can recieve correctly from/with other libraries for ios and Mac 

we have tried also this 

You could also try out changing GattWriteOption::WriteWithoutResponse to GattWriteOption::WriteWithResponse at https://github.com/adabru/BleWinrtDll/blob/main/BleWinrtDll/BleWinrtDll.cpp#L536 . That would make the call wait for the BLE device to send a "I received" confirmation. In BLE terms that would be "request"+"response" instead of a "command" without a response.

but the returnis still false.

what can we do about it?

Thank you in advance.

adabru, 2023-12-15:
Hi @tommasoFabbri :),
please open a new issue and write which error message you are getting. Please also try the Microsoft provided sample linked in the Readme to checkout if sending data works there.