adabru/BleWinrtDll #22: Have You added the Ability to Disconnect() ?

Ugobyte, 2022-01-16:
This API is really great and I actually won't mind financially supporting you. Where can I donate?

I would like to know if you have or plan to add a Disconnect() feature.

thangdhz, 2022-02-07:
Add this function to `BleWinrtDll.cpp`
```
void Disconnect(wchar_t* deviceId) {
	for (auto device : cache) {
		if (hsh(deviceId) == device.first) {
			device.second.device.Close();
			for (auto service : device.second.services) {
				service.second.service.Close();
			}
			cache.erase(hsh(deviceId));
			return;
		}
	}
}
```
Add this function to `BleWinrtDll.h`
```
__declspec(dllexport) void Disconnect(wchar_t* deviceId);
```

Add this function to `BLE.cs`

```
        [DllImport("BleWinrtDll.dll", EntryPoint = "Disconnect", CharSet = CharSet.Unicode)]
        public static extern void Disconnect(string deviceId);
```

Ugobyte, 2022-02-08:
@thangdhz Thank you, this is great!
Do I have to recompile a new .dll if i add ```void Disconnect(wchar_t* deviceId)``` to the .dll file?

thangdhz, 2022-02-09:
yes, need recompile .dll file

Ugobyte, 2022-02-17:
> yes, need recompile .dll file

Excellent, Thanks

adabru, 2022-04-25:
Hi @Ugobyte , thanks for the offer! I probably can't work on this project for like the next ten months. You can feel free to hire freelancers to fix some of the most popular open bugs 😉 or maybe sponsor one of the forks. I would gladly redirect to it if it gets more features.

BtuApple, 2022-12-16:
@thangdhz This code works, but do not work well. After disconnect from device, you can not find the same device by scan function.Is there something wrong with the scan codes?

ahSOLO, 2023-05-05:
Thanks, this code worked for me