adabru/BleWinrtDll #26: Write to FTMS Device

serramat, 2022-02-23:
Hi,
I am using your library to read the data from the BLE sensors and I was able to receive the information correctly.
Now I would also need to write data on some devices (such as assigning the clutch to a roller). I am using the WritePackage but I can never write! I also tried your example but without any success.

To be precise I'm trying to write on the 00002ad9-0000-1000-8000-00805f9b34fb characteristics of the 00001826-0000-1000-8000-00805f9b34fb service

Thank you so much.
Matteo.

adabru, 2022-03-20:
Hi @serramat , you could try the things mentioned in https://github.com/adabru/BleWinrtDll/issues/20#issuecomment-898939454 .

adabru, 2024-01-12:
Hi @ShuohengZhang ,
it's good that your UWP application works. The dll in this repository is a c++ winrt wrapper for UWP code so the step from your UWP app to the dll should be small. 

You can compare the UWP function calls from https://github.com/adabru/BleWinrtDll/blob/main/BleWinrtDll/BleWinrtDll.cpp with your UWP app. If your UWP app uses another API-call, you can change that in BleWinrtDll.cpp. Or you can change the API-call in your UWP app to the same as in BleWinrtDll.cpp.

As reference for converting UWP code to c++ winrt and vice versa, you can use
https://github.com/microsoft/Windows-universal-samples/blob/main/Samples/BluetoothLE/cppwinrt/Scenario2_Client.cpp
and
https://github.com/microsoft/Windows-universal-samples/blob/main/Samples/BluetoothLE/cs/Scenario2_Client.xaml.cs
Those are functionally the same.

ShuohengZhang, 2024-01-13:
Hi @adabru,
Thanks for your reply!! And yeah, I was a bit afraid to write a C++ DLL by myself because I don't really understand the C++ codes that well. However, I tried it with your suggestions today and looked into your codes and the samples I made ([Dll_BLE.cpp](https://github.com/ShuohengZhang/DLL_BLE_BIKE/blob/main/Dll_BLE.cpp)). When I import this DLL into the Unity project, there is a DllNotFoundException when I already put it in the same folder as your BLEWinrtDll (I assume when this DLL can be used, so should mine). I would so appreciate hearing your suggestions!! 

Best regards!

adabru, 2024-01-13:
I'm not quite sure what the reason for the DllNotFoundException is. You can try to look at Unity's import settings for that file. You can also try to change the 32-bit or 64-bit setting in VS. And maybe try to ask ChatGPT.

Is your repository private? I can't access the link.

ShuohengZhang, 2024-01-14:
@adabru sorry I didn’t notice that was private now u may have the access. And thanks again that you still concern about my case !