adabru/BleWinrtDll #27: Clarification of Readme with regards to unity editor

Abrahamh08, 2022-03-14:
I think this statement needs clarification:
"With Unity's future integration of C# version 5 this repo most probably should become obsolete as C# version 5 is meant to merge .NET-Framework and .NET-Core so you should be able to call Windows' BLE API directly from C# then."

https://docs.unity3d.com/Manual/windowsstore-scripts.html notes that the Unity Editor always uses mono. That means that while you can build your application with calls to the Windows SDK, you cannot test it in the editor. This is important because when building for platforms like Android, the entire purpose of integrating BLE for windows is just to test the application in the editor. Therefore, I do not think this repository can ever become obsolete, as people will need to maintain it as an interface for the editor. Am I wrong?

adabru, 2022-03-20:
Hi @Abrahamh08 , I think you are right. I thought the editor's incapability of calling .NET-Core code was the main problem. But as you pointed out, the editor runs on mono. That should prevent calling windows API even when it will run on Core. Do you mind creating a pull request?

Abrahamh08, 2022-03-21:
Sure. #28

closed 2022-03-21 by Abrahamh08