adabru/BleWinrtDll #29: Disconnect Event Not Implemented

Roamer79, 2022-04-24:
Really needs an event for disconnections as you could be left in the dark. Otherwise a good start.

Quget, 2022-04-25:
> Related: #15 , #6
> 
> @nathanrona @Quget did you make any progress into this direction that you can share with Roamer79 ?

Hi, I didn't explore the BleWinrtDll further. It wasn't neccesary for my project anymore.

Roamer79, 2022-04-26:
> Related: #15 , #6
> 
> @nathanrona @Quget did you make any progress into this direction that you can share with Roamer79 ?

Hi Adabru, how about I pay you for you time to implement this for us? I've looked into it and its above my ability! I don't know how to wrap that function

adabru, 2022-04-26:
Hi Roamer79, thanks for the offer! But I can't, see https://github.com/adabru/BleWinrtDll/issues/22#issuecomment-1108535578 . If you wanna try to implement it yourself I could try to assist you with that.