adabru/BleWinrtDll #32: Multiple devices support

tufeixp, 2022-05-05:
Does this lib support a multiple devices running?

adabru, 2022-05-05:
I think yes but I didn't try. See https://github.com/adabru/BleWinrtDll/issues/7#issuecomment-771600983: 
> I guess each bluetooth device normally has a unique device id. In that case you should be able to just call the functions with different device-ids. Each received data-package contains information about the device it was sent from. Likewise, in each data-package you send you can specify the device-id it should go to.

tufeixp, 2022-05-06:
Yes, I tried and it works~ thanks~

closed 2022-05-06 by tufeixp

mnajib76, 2022-08-30:
Hi @tufeixp , can you share your code/project how you read the data when connect to more than 1 device?

mnajib76, 2022-08-30:
[image]

So based on my understanding we need to SubscribeCharacteristic() for both devices id , services and char id, and all buffer will still pull from the function as in picture above?

adabru, 2022-09-01:
Hi @mnajib76, you are right. You pull from the function and determine which device it was sent from by inspecting `res.deviceId`. https://github.com/adabru/BleWinrtDll/blob/31fcc9fd733e16a80d712ece893caf0c678f9111/BleWinrtDll/BleWinrtDll.h#L22-L28

mnajib76, 2022-10-15:
Thank you very much! Manage to do it!