adabru/BleWinrtDll #34: Device not connecting after scanning

AmalPBanerji, 2022-08-11:
Tried "build and run" from unity with these settings,
Architecture - Intel 64-but
Build type - D3D project
Target SDK version - Latest installed
Min Platform version - 10.0.10240.0
VS Version - latest installed (2019)
Build and run on - local machine
Build config - Release
copy references - untick
copy pdb files - untick
development build - untick
IL2CPP code gen - faster runtime
compression method - default

After running the application, scanning pressed, device selected, when scan for services given, " Failed to connect to device"  error occurs.

adabru, 2022-08-14:
For investigation, you can download and run the [Microsoft's Bluetooth Low Energy sample](https://docs.microsoft.com/en-us/samples/microsoft/windows-universal-samples/bluetoothle/) and see whether scan for services works there. This library is like 97% a clone of that sample. 

Does it never work, or does it work sometimes?

SightlessDog, 2022-09-02:
Same happens to me rn but on the hololens 2, in the editor everything works fine but when I try it on the hololens I get this error 
"Failed to connect to device"

adabru, 2022-09-03:
@SightlessDog, maybe the linked project in https://github.com/adabru/BleWinrtDll/issues/23#issuecomment-1035010515 can somehow help you? Though I'm not sure whether you have the same problem.

SightlessDog, 2022-09-07:
@adabru thank you for the answer, well I had to go through that step, but the problem I had kind of got fixed from itself (I uninstalled the app and installed again after making a force rebuild )