adabru/BleWinrtDll #37: How can i  subscribe to multiple BLE characteristics?

damhirgomez, 2022-09-18:
first for all, thanks for this DLL, is amazing, but i need to suscribe to multiple characteristics from a xiao sense ble, can you help to know how can i made this? sorry for my english and thanks for your reply.

adabru, 2022-09-27:
Hi @damhirgomez , you can call

```cs
BleApi.SubscribeCharacteristic(selectedDeviceId, selectedServiceId, selectedCharacteristicId, false);
```

for each desired characteristic and check the `res.uuid` after the `BleApi.PollCharacteristic(out res, false)` call.

damhirgomez, 2022-10-25:
Thanks 4 ur help!, really works, i have other question, how can i do a reScan?, in the first scan for some reason the Name of the Ble devices is = "", so i need to close the app and start again to see devicesName, thank again