adabru/BleWinrtDll #39: After called Quit()，Can not cann't find connectedable devices.

BtuApple, 2022-12-15:
I am doing reconnect works.After called Quit() function, I found the scan func didm't work. I guess it is because the filter in c++ code worked in error.
`	IVector<hstring> requestedProperties = single_threaded_vector<hstring>({ L"System.Devices.Aep.DeviceAddress", L"System.Devices.Aep.IsConnected", L"System.Devices.Aep.Bluetooth.Le.IsConnectable" });
` 
I need connect and disconnect.So, what can I do?

adabru, 2022-12-16:
Do you get any error message in the bottom-left corner of the window?
Does a rescan work before calling Quit()?
As an alternative, maybe subscribe and unsubscribe are sufficient?

BtuApple, 2022-12-22:
I think there is something wrong with the c# code.
1. when I want next scan, the cancel flag is still true, I must reset it
2. when I do scan、connect、disconnect for loops, it crashed after sometime,because you called Quit() in destructor function.
I fixed it in my own code, now it works all right. C++ code is very good, no error.