adabru/BleWinrtDll #40: How to make it work on Android?

userqt, 2023-01-21:
Hi,

thank you for the great example. I tried it an it works quite well in the editor. I was wondering if this can be made to work for Android and how?

adabru, 2023-01-22:
Hi @userqt,

this library doesn't work on android. You have to additionally use another library. 

You can check out the asset store: https://assetstore.unity.com/?q=Bluetooth&orderBy=1 . Many of those also work on iOs.

If you want a open source solution you can check out the active project https://github.com/Velorexe/Unity-Android-Bluetooth-Low-Energy or the archived project https://github.com/seiji/unity-bluetooth . That's at least what I found after a google search.

userqt, 2023-01-22:
Thank you, I will take a look.

adabru, 2023-01-22:
Actually there is already an issue for android support: https://github.com/adabru/BleWinrtDll/issues/12
So closing this as duplicate. Please feel free to comment on the other issue's thread 👍

closed 2023-01-22 by adabru