adabru/BleWinrtDll #41: Connection with multiple devices

EAriu, 2023-01-28:
Hi, thank you for this amazing repository. I was using your code to connect to a single custom device without problem for a project. In particular, I modified it using threads so that I could receive packages and in the meantime be able to do other things with the Unity application. Now I want to connect multiple devices at once. Is it possible? I tried with multiple instance of the same class but they seem to interfere with each other. 

Since I don't have much experience in programming, I tried momentarily to create two different classes that utilize two different dlls. In this way I seem to be able to receive packets correctly from two devices at the same time. However, this method is not sustainable in case I want to connect more than two devices. Is there a way to connect to more than one device without using the gimmick I am using right now? Is it possible that I am missing something and is it easier than it seems?

Thank you!

adabru, 2023-01-30:
See here: https://github.com/adabru/BleWinrtDll/issues/32

EAriu, 2023-01-30:
I hadn't thought that BLE data contains also the device Id! Thank you very much, I'll try to do it that way!

closed 2023-01-31 by EAriu