adabru/BleWinrtDll #42: Missing DLL after Build

eproasim, 2023-02-06:
The readme mentions that this dll should work for a Windows Standalone build, but I have been unable to actually produce a working build of this project or my own which was was built with the existing BleApi.cs script. I am not terribly familiar with dll's in unity, and the available answers around the web are unclear to me. On each frame, I receive a DLLNotFoundException immediately at startup after building.  This is not an issue when using the Unity editor. Could anyone shed light on what I'm doing wrong?

eproasim, 2023-02-06:
Solved embarrassingly quickly after posting this. For anyone else that might have this problem. Please ensure that your application requires the latest Microsoft Visual C++ Redistributable to be installed.

closed 2023-02-06 by eproasim

adabru, 2023-02-11:
Thank you eproasim 🙂, I added it to the readme.