adabru/BleWinrtDll #47: Connecting and Subscribing to heart sensor

99bits, 2023-04-04:
Thank you for this really amazing project.  
I have a heart sensor and it connects fine and the Scan services and Scan characteristic seems to populate fine.
When I try to subscribe to the sensor, the editor completely hangs.  I also built the app and tested it.  It also hanged when trying to subscribe  to the data.  
[image]

Do you know what I am missing.

Thank you in advance.

adabru, 2023-04-09:
I have no clue. You can try this out: https://docs.microsoft.com/en-us/samples/microsoft/windows-universal-samples/bluetoothle/ and report whether that works for the heart rate measurement subscription.

lazerbots, 2024-07-13:
Hey, I am new to BLE programming and was just checking out the issues. I did notice that in this screenshot the devices appear dozens of times again and again. When I was checking out your BleWinRRDll example project I had the same "problem". I was wondering if this is normal or how I can avoid this. Thanks!

adabru, 2024-07-14:
Thanks @lazerbots for the remark! You're right, the lines https://github.com/adabru/BleWinrtDll/blob/03b4dcf71a4152958074b90cb5459cf575456bb7/BleWinrtDll%20Unity/Assets/Scenes/Demo.cs#L67-L74 don't check whether the device was there before. That's a bug. If the device sends multiple updates and has IsConnectable to true on some of them, it will be listed multiple times.

I don't know if that could have something to do with the hanging.