adabru/BleWinrtDll #48: Unable to write attribute

StevenDextrain, 2023-04-24:
Hello !
First of all, thanks a lot for your project and the work you've done, as a novice when working with Bluetooth, it helps a lot. I've scoured the different issues already listed on this repo, but can't find something that answers my questions.
That being said, I'm currently having an issue when trying to subscribe to my write notification on my bluetooth device.
I get this error :
"Could not establish connection to device with ID BluetoothLE#BluetoothLEc4:23:60:eb:a3:ec-c5:02:be:bb:2a:69
System.Exception: Connection failed: BleWinrtDll.cpp:501 SubscribeCharacteristicAsync catch: Unable to write attribute **(translated from French, not sure about the translation, sorry about that)**"

The strange thing is that it happens when trying to subscribe to my device's Write characteristic.
I have 2 different characteristics on my device, one that is for Read/Notify, and one that is for Write, and I figured the script would be able to identify which is which, and subscribe to notifications for the Notify one, and subscribe to the Write characteristic to send messages, but I realize I might be wrong ?

I also had another question, it might be because I can't reach this part yet because of my previous issue, but I don't see a way to read data sent by the BLE device, am I missing something ? I looked for "Read" "Receive" and other keywords in the code but found no results

Anyways, I'm very much stuck and I'd really appreciate if you could help me 😁
Thank you so much, have a nice day !

adabru, 2023-04-24:
Hi @StevenDextrain :-)

you only subscribe to characteristics that issue notify events. After subscription you can read incoming messages with PollData.

You can write to characteristics without prior subscription.

"ReadCharacteristic" is not implemented in the dll although it is a ble functionality. If you need it you can checkout https://github.com/adabru/BleWinrtDll/issues/6#issuecomment-1480375448 . But subscribing alone might be sufficient for you.

StevenDextrain, 2023-04-25:
Hey again !
Thanks for so quick of a response 😁
I'm wondering what you mean by "But subscribing alone might be sufficient for you.", is there already a way to collect my device's data without implementing ReadCharacteristic ?

StevenDextrain, 2023-04-26:
Ah nevermind, I just realized I missed part of the first (and most important part of your reply 🙃), I'm gonna try that and come back with another question, or hopefully, everything will work fine and I'll close the issue !
Thanks again !

adabru, 2023-05-13:
> BleApi.SendData(in data, false) method call returned "false".

For the non-blocking call, the result is always false:

https://github.com/adabru/BleWinrtDll/blob/f2635b7b02432ea3006983657107f092c862171d/BleWinrtDll/BleWinrtDll.cpp#L554-L560

You should see an error in the message window eventually, if it doesn't succeed. The blocking call returns true on success of sending:

https://github.com/adabru/BleWinrtDll/blob/f2635b7b02432ea3006983657107f092c862171d/BleWinrtDll/BleWinrtDll.cpp#L537-L540

 The success doesn't guarantee that the package arrives at the target device though. There is a possibility to do that though.

> PollData is used to read what the device is sending constantly, right ?

Exactly:

**SubscribeCharacteristicAsync**

https://github.com/adabru/BleWinrtDll/blob/f2635b7b02432ea3006983657107f092c862171d/BleWinrtDll/BleWinrtDll.cpp#L492

**Characteristic_ValueChanged**

https://github.com/adabru/BleWinrtDll/blob/f2635b7b02432ea3006983657107f092c862171d/BleWinrtDll/BleWinrtDll.cpp#L478

**PollData**

https://github.com/adabru/BleWinrtDll/blob/f2635b7b02432ea3006983657107f092c862171d/BleWinrtDll/BleWinrtDll.cpp#L520-L524