adabru/BleWinrtDll #53: BleScan.Impl.PollData make my Unity crashed.

AlvinLBS, 2023-08-17:
Hi, im a student doing a FYP about BLE communication. Sorry that I only had very little knowledge about Unity and Visual studio coding.

I tried to debug line by line, notice the Unity crashed when it reached PollData. So i try to search the error in the library and notice is the QuittableWait(dataQueueSignal, lock) make my Unity crashed. I understand PollData is a function to retrieve data from BLE device, and QuittableWait() used to wait unitl there is data. My BLE device characteristic did not have data thats why it keep waiting (might causing loop).

I create a backup and try to ignore QuittableWait() function and replace it with saveError. So when block=true, and data is empty, it will saveError "no data". Everything work fine and Unity also did display the error (i modified Demo.cs and BLE.cs too so I can start ReadPackage when click subcribe). 

May i know what is the problem?

AlvinLBS, 2023-08-17:
I noticed if I run the Program.cs, when i hit the Impl.PollService, my Unity also crashed. PollService also using QuittableWait( ) function.

adabru, 2023-08-19:
Hi Alvin :),
do I understand correctly:

- You run the unmodified Demo.
- The device-, service- and characteristic-scan all work fine.
- When you select a characteristic and click "Subscribe", Unity crashes at following line:

https://github.com/adabru/BleWinrtDll/blob/f2635b7b02432ea3006983657107f092c862171d/BleWinrtDll%20Unity/Assets/Scenes/Demo.cs#L131