adabru/BleWinrtDll #55: Does BleWinrtDll provide API for pairing with passkey?

snowuyl, 2023-10-24:
I would like to implement pairing with passkey on Windows 11. Does BleWinrtDll provide API for pairing with passkey?

adabru, 2023-10-24:
I did a googling and found [a question on SO](https://stackoverflow.com/q/57710935/6040478) regarding UWP pairing PINs and a [microsoft sample](https://github.com/microsoft/Windows-universal-samples/tree/main/Samples/DeviceEnumerationAndPairing) with C# and cppwinrt . BleWinrtDll does not provide pairing with passkey. This is the implementation of BleWinrtDll: https://github.com/adabru/BleWinrtDll/blob/main/BleWinrtDll/BleWinrtDll.cpp

This is the exposed API: https://github.com/adabru/BleWinrtDll/blob/main/BleWinrtDll/BleWinrtDll.h

But looking at the microsoft sample, it should be possible to extend BleWinrtDll with passkey support. You could try to download and run the linked microsoft sample. If it works out, you can transfer its code to BleWinrtDll and use it in Unity. It should work.

snowuyl, 2023-10-25:
Thanks for your reply! I would like to implement a Windows C++ application (I.e. not Unity application). Do you suggest to use BleWinrtDll?

adabru, 2023-10-25:
If your outside Unity, things get easier. Just use the Windows API without BleWinrtDll. I might suggest C# instead of C++ as I perceive C# as faster to program than C++. But that's just preference.

snowuyl, 2023-10-25:
Thanks for your reply!

closed 2023-10-25 by snowuyl