adabru/BleWinrtDll #57: Write to device problem

tommasoFabbri, 2023-12-18:
Hi,
we started to use the BlewinrtDll we need it for our Unity app and the solution seems great.

we can connect to the BLE device and can receive data but the function "SendData" always return false, with no error message 

we have try to use the precompiled dll that we find in the unity example non threaded
the device can recieve correctly from/with other libraries for ios and Mac

we have tried also this by recompiling the dll 
--
You could also try out changing GattWriteOption::WriteWithoutResponse to GattWriteOption::WriteWithResponse at https://github.com/adabru/BleWinrtDll/blob/main/BleWinrtDll/BleWinrtDll.cpp#L536 . That would make the call wait for the BLE device to send a "I received" confirmation. In BLE terms that would be "request"+"response" instead of a "command" without a response.
--

but the returnis still false.

also we don't have the right environment to compile the Microsoft example

what can we do about it?

Thank you in advance.

adabru, 2023-12-25:
Hi @tommasoFabbri ,

> also we don't have the right environment to compile the Microsoft example

Maybe you can elaborate on that?

> the function "SendData" always return false, with no error message

That is expected for the nonblocking call. You can try to set `block` to true.

tommasoFabbri, 2023-12-27:
think i manage to solve the problem but now in windows build with the dll built for win32 i have a crash when i start to read from ble device.
I get this error:
"The thread tried to read from or write to a virtual address for which it does not have the appropriate access"