adabru/BleWinrtDll #58: Error subscribing to Characteristic

arafi123, 2023-12-27:
Hello,
I am trying to connect to BLE Weight scale (Service 0x181D). When I subscribe to characteristic , I get following error:

`BleWinrtDll.cpp:501 SubscribeCharacteristicAsync catch: The attribute requires authentication before it can be read or written.`

How can I fix this ? What needs to added as authentication parameter in code so that the data can be read from the device ?

I look forward to your support.

Kind regards

adabru, 2023-12-31:
Hi @arafi123 , maybe you need to pair the device. Maybe related: https://github.com/adabru/BleWinrtDll/issues/7

arafi123, 2024-01-02:
You mean pair it with PC ? For this BLE Weight scale, I get the same authentication error also with the Microsot BLE sample so I assume may be the device is encrypted or something.

adabru, 2024-01-07:
Yes, I meant pairing it with the PC. Did you try that? I think you can do it via Windows Settings, I'm not  sure.