adabru/BleWinrtDll #64: Microbit Bluetooth UART service data reading problem

nmitrak, 2024-04-03:
Hi @adabru, as i wrote in a previous post i managed to connect unity and then hololens 2 with a microbit and read its internal temperature sensor value by subscribing to the respective ServiceId and CharacteristicId. But now i have a second temperature sensor that i have connected to a microbit's pin and wrote a program for the microbit in order for the temperature value of the sensor (a simple number) to be transmitted though the Bluetooth Uart service which i enabled on the microbit.
Although i managed to send a set of characters from Unity to the microbit , through your BlewinrtDll project , by using the Demo's Write function , which uses BleApi.SendData,  i cannot read any data coming from the microbit. It seems to me that when unity polls data from the CharacteristicId {6e400002-b5a3-f393-e0a9-e50e24dcca9e} of the Bluetooth UART Service nothing comes out. Subscribing to this service doesnt seem to detect value changes when using your demo(?) 
The funny thing is that i can read the values coming from microbit through this bluetooth Uart service when using an android phone ,with a custom made application with appinventor  and that the Microsoft 's Sample BluetoothLE app from Hololens 2 can subscribe to this CharacteristicId and seems to be detecting value changes but cannot recognize the data format.
I would appreciate any ideas . Again thank you very much.

nmitrak, 2024-04-04:
After some research it seems that for that particular bluetooth uart service microbit sends raw binary data. Maybe unity polls for incoming data that are not in raw binary format

adabru, 2024-04-05:
Hi nmitrak, binary format should be fine for BleWinrtDll. In the Microsoft sample, you can try setting a breakpoint or a log output at https://github.com/microsoft/Windows-universal-samples/blob/main/Samples/BluetoothLE/cs/Scenario2_Client.xaml.cs#L447

nmitrak, 2024-09-22:
Thank you again for your answer. Just coming back again to the same  problem after a long period of time. I read somewhere that in order to receive messages from the bluetooth uart service of Micro:bit you have to enable notification, writing "0200" on the Client Characteristic Configuration Descriptor (CCCD) of the UART Service Characteristic.(The CCCD is 2902). Is this possible with BleWinrtDLL you compiled?
I would like to thank you once more for your contribution in solving this problem.

adabru, 2024-09-24:
Subscribing to a characteristic writes to the CCCD:

https://github.com/adabru/BleWinrtDll/blob/03b4dcf71a4152958074b90cb5459cf575456bb7/BleWinrtDll/BleWinrtDll.cpp#L482-L490

nmitrak, 2024-09-28:
Dear @adabru  thank you very for your answer and for clarifying this point. Nevertheless microbit is sending data through bluetooth uart and HoloLens doesn't retrieve them when subscribing to a characteristic id and polling data. I tested the communication between microbit and Hololens 2 with Microsoft's bluetooth sample and data comes in but of unknown format.  That's why i thought of implementing the ReadDataAsync function as you suggested at an older post. But being a little bit of a newbie at this made me wondering how this function works as it doesn't seem to take any account of descriptors
Again thank you very much for your time

adabru, 2024-11-14:
Thank you very much for sharing your solution :)

closed 2024-12-15 by nmitrak