adabru/BleWinrtDll #66: Can't write the correct values

Dragelans, 2024-06-02:
Hi, I have a BLE balance board I want to connect to Unity but having issues when writing to the device.
I tested with Universal Sample and some other libraries. The Device is working and also can get notify from the Unity Demo scene.

As example one of the characteristics control the led on the board as R G B Rainbow and gets 4 byte value so when I send "FF 00 00 00" it will glow red.
I tried to change parse text input as Hex but couldn't manage to write to the board.

```    
public static Byte[] FormatData(string data, DataFormat format)
    {
        try
        {
                string[] values = data.Split(new char[] { ' ' }, StringSplitOptions.RemoveEmptyEntries);
                byte[] bytes = new byte[values.Length];

                for (int i = 0; i < values.Length; i++)
                    bytes[i] = Convert.ToByte(values[i], (format == DataFormat.Dec ? 10 : (format == DataFormat.Hex ? 16 : 2)));

            return bytes;

            
        }
        catch (Exception error)
        {
            return null;
        }
    }`

`
    public void Write()
    {

        byte[] payload = FormatData(writeInput.text,DataFormat.Hex);
        BleApi.BLEData data = new BleApi.BLEData();
        data.buf = new byte[512];
        data.size = (short)payload.Length;
`

adabru, 2024-06-03:
Hi @Dragelans :),
what error message do you get?

Dragelans, 2024-06-03:
Hi @adabru ,

I don't get any error , when I set blocking mode true it just freezes though so not sure if it's totally cause of an error or the format type I send

Dragelans, 2024-06-07:
Hi @adabru again, 

With the dll, I wrote a test console application to check blocking mode. It returns "Write Success"  but having no affect on the device. 

In datasheet, there are specific lengths in this case 4 bytes. on [toolBLEx](https://github.com/emericg/toolBLEx) I manually send the bytes it works but if send more than 4 it doesn't have any affect so I thought it might be about the buffer size maybe but when i change buffer size to bytes length it gives error about not correct size and closes the application.

adabru, 2024-06-08:
Are you setting the BLEData size property to 4?

Dragelans, 2024-06-08:
I did, but on code when sending it's already resizing bytebuffer array from 0 to data size so that was not the issue.

I checked another BLE library for console and compared the send code. there were 2 main differences.

1- Byteorder, I see in reference library it goes something like "Writer.Byteorder(LittleEndian)" meanwhile it didn't on dll. I actually encountered this in another library where i had to enter bytes reverse. Though don't think that was the main problem since even if I entered in wrong order it should have done something.

2- **WriteWithoutResponse** , in other library It was **WriteWithResponse** so I changed that and now it's working. Since haven't work on bluetooth before I have no idea why it makes a difference but now it works.

https://github.com/adabru/BleWinrtDll/blob/f2635b7b02432ea3006983657107f092c862171d/BleWinrtDll/BleWinrtDll.cpp#L533-L536

adabru, 2024-06-10:
Thank you very much for sharing. There seems to be a related issue: https://github.com/adabru/BleWinrtDll/issues/33

Do you have a thought on how that could be implemented so that users won't have the same hard time as you did? Otherwise it can be added to the readme.

Dragelans, 2024-06-11:
> Thank you very much for sharing. There seems to be a related issue: #33
> 
> Do you have a thought on how that could be implemented so that users won't have the same hard time as you did? Otherwise it can be added to the readme.

Good question, though not sure. Seems like difference between With and Without Response is With sends back acknowledgment but don't know how this affects the device since again as I mentioned, don't have much experience with bluetooth in general.

I'd say WithResponse can be added as default to repo but that might slow down the connection as waiting response causes a bit delay.  

Maybe a section on readme can be added for possible errors and solutions

commit 03b4dcf7 (2024-06-13): Update readme.md (closes #66)
closed 2024-06-13 by adabru