adabru/BleWinrtDll #69: Access is Denied Error

justindpnt, 2025-03-21:
Hello! Love the dll, great idea. Except I am running into one problem. I can connect to my Arduino alright, but if I try to write using the sample scene I get an "Access is denied." error. Exact error...

BleWinrtDll.cpp:545 SendDataAsync catch: Access is denied.
UnityEngine.Debug:LogError (object)
Demo:Update () (at Assets/Scenes/Demo.cs:143)

Any idea as to why this could be happening? I was able to connect from my phone and successfully write to the Arduino, but doing the same thing from windows has been really annoying. This is probably my 5th solution I have tried...

Can confirm it connects, find the attribute and characteristic all fine, just runs into an access is denied message when trying to write.

adabru, 2025-03-22:
Hi @justindpnt :),

try pairing the device from Windows settings. If that doesn't help, you can try the debugging steps from https://github.com/adabru/BleWinrtDll/issues/7#issuecomment-770755841.

Related: https://github.com/adabru/BleWinrtDll/issues/58

justindpnt, 2025-03-23:
Hey @adabru! Thanks for replying so fast.

So I looked into it, first - I am using an arduino Uno R4, so I have BLE but not actual bluetooth. Trying to connect to the computer with regular bluetooth throws an error. So unfortunately I can't go that route.

I did download the Microsoft .net sample, and I was able to see, connect, and write to my arduino. I can do the same with a mobile application, and with another windows BLE testing application. So it IS able to work somehow, but then trying to do so from Unity throws that permission error. Is there some setting I'm missing that I have to enable?

[image]

I guess I can look into getting an adafruit esp board that supports both bluetooth and ble so I can connect to Windows first... seems like that is what solved it for others.

adabru, 2025-03-23:
If the Microsoft sample works, the DLL approach is quite certain to work with some modifications as well :)

You can further try changing `GattWriteOption::WriteWithoutResponse` to `GattWriteOption::WriteWithResponse`. See https://github.com/adabru/BleWinrtDll/issues/20#issuecomment-898939454.

I'm not sure how pairing/bonding/connecting works with BLE. There seemed to be some issues where pairing/connecting seemed to help. But because the Microsoft sample works, you don't need to look further into it 👍 .

To use the sample for modifying the DLL, you would do:

- Use the sample's  [C++ winrt code](https://github.com/microsoft/Windows-universal-samples/blob/main/Samples/BluetoothLE/cppwinrt/Scenario2_Client.cpp) or [C# code](https://github.com/microsoft/Windows-universal-samples/blob/main/Samples/BluetoothLE/cs/Scenario2_Client.xaml.cs) as reference. If you downloaded the sample, you can inspect your downloaded code instead. You can also try to make changes in the sample-code to get more insights.
- Modify BleWinrtDll.h, BleWinrtDll.cpp and BleApi.cs as needed. In your case only changes in BleWinrtDll.cpp should be sufficient.
- Rebuild the DLL and check it out in Unity.
- For faster iterations you can use the DebugBle program which is included in this repository.

justindpnt, 2025-03-23:
Hm okay well good to know. Do you happen to have a fork or copy of the DLL with the WriteWithResponse change? I've been trying to build a new version for a few hours now but haven't had any luck. I think new visual studio version + library version differences is causing lots of hiccups in the build process.

Appreciate it! :)

adabru, 2025-03-23:
You can try out jadware's fork: https://github.com/adabru/BleWinrtDll/compare/main...jadware:BleWinrtDll:main .

If you manage to use this repository, feel free to create a PR for the VisualStudio update. If you replace the API `SendData` with `WriteWithResponse` and `WriteWithoutResponse` that would make an amazing PR as well :)

Personally I switched to building an .exe from C# code for Windows + an .aar from kotlin code for faster debug iterations. It's very similar to how https://github.com/Velorexe/Unity-Android-Bluetooth-Low-Energy is working for Android. It's tighter integrated with hardware specifics and not suited for generic use because of that, though.