adabru/BleWinrtDll #70: BleWinrtDll.cpp:429 ScanCharacteristicsAsync catch: 该对象已关闭。

Amy2048, 2025-03-31:
It is impossible to obtain the Characteristic and an error message "BleWinrtDll.cpp:429 ScanCharacteristicsAsync catch: This object has been closed." is reported.

Amy2048, 2025-03-31:
After the first successful subscription to the feature, closing the application Bluetooth is not disconnected, and running the software again will not be able to obtain characteristicUuid

closed 2025-04-01 by Amy2048

Sola-AIGithub, 2025-05-05:
hello, do you solve this problem?

Amy2048, 2025-05-05:
> 您好，您解决了这个问题吗？

I've fixed the issue, added a method for active disconnect.

Sola-AIGithub, 2025-05-05:
May I ask if you are using BleApi.Quit() to do a buffer clear and then reconnect the device, then scan its service & characteristic?
I found that when my device was disconnected, I was able to search for the service but not the characteristic when device reboot.
Since I'm using a one-to-many device connection, I don't want to use BleApi.Quit() to clear the entire buffer!

I'd appreciate any help here.

Amy2048, 2025-05-06:
> May I ask if you are using BleApi.Quit() to do a buffer clear and then reconnect the device, then scan its service & characteristic? I found that when my device was disconnected, I was able to [...]

I am using the DLL from the https://github.com/Joelx/BleWinrtDll-Unity-Demo.git library and not the current library's DLL. When I need to exit, calling Quit() disconnects the connection.

Sola-AIGithub, 2025-05-07:
I've study BleWinrtDll-Unity-Demo project, it seems like once you call Close(), it will run Quit() to clean all the buffer in BLE connection...
But thanks you help!

adabru, 2025-05-17:
Hi @Sola-AIGithub, thank you very much for posting your solution! The link to your comment is now added in the FAQ of the readme.